levylabpitt/Instrument-Framework #5: Generalize Configuration subsystem

ciozi137, 2020-05-18:
[image]

ciozi137, 2020-05-18:
https://drive.google.com/file/d/1HXjSoFdJEa7-5yU6IP98QrxBk5HNRNSk/view?usp=drivesdk

ciozi137, 2021-09-24:
Incremental configuration architecture & overrides:
v1.9.3:
[image]

next release (v1.9.4):
[image]

ciozi137, 2022-04-27:
@Monkeymerlot I need to revisit this configuration stuff for the Instrument Framework. Did you ever come up with something for the Litho software that you are happy with?

Monkeymerlot, 2022-04-28:
@ciozi137 I remember working on something for a while and now I can't find it. I'll keep looking once my finals are over (ie on Monday). I didn't see it in my recycle bin (which has never been cleaned out) I saw a little bit of stuff in the Maui Framework Repository but it doesn't look like what I think you are talking about.

For the Lithography software I have just been doing real basic write to ini files (using the configuration Read Write VIs in the Instrument Framework). In the future I want to add different folders for Users, and each user would have separate configurations.