levylabpitt/Instrument-Framework #32: Read Multiple Section Clusters or Modified Section Clusters

Monkeymerlot, 2021-02-17:
https://github.com/levylabpitt/AFM-Lithography/issues/16#issuecomment-780109581
https://github.com/levylabpitt/AFM-Lithography/issues/16#issuecomment-780122280
https://github.com/levylabpitt/AFM-Lithography/issues/16#issuecomment-780195806
So I thought about the scenarios that we could have pertaining to the issue that I had in the other project and laid out what I thought would be the best way to go about this.
[image]

Replacing the default data with the existing data from file when the cluster size has grown (ie the section is missing an element) was complicated greatly by the fact that the returned data from the read ini file VIs does not have labelled elements. 
Here is the test setup that I did:
1. I created a file and initialized it with the following content:
```
[Test Cluster]
Hello = "World"
Number = "1"
Number 2 = "3"
```
2. Then I requested that it read from the file a cluster containing 1 more element than what was in the file. To make sure that the variant to data wasn't coercing the data I changed the values in the type cluster before I used it in variant to data:
[image]

3. This is what I came up with that correctly wrote to the file and returned the correct values:
[image]
And the file contents:
[image]
returned data:
[image]

Files:
[Configuration Prototype.zip](https://github.com/levylabpitt/Instrument-Framework/files/5998533/Configuration.Prototype.zip)

ciozi137, 2021-02-24:
@Monkeymerlot thanks I will think about this. After first read can we just test if variant is empty?

Monkeymerlot, 2021-02-24:
@ciozi137 Good question. I will check but I have a suspicion that it will return something like this:
```
" " =
" " =
" " =
```
Also I think that the fact that there are no names on the returned cluster is actually a bug in the openG VIs. I am thinking of opening an issue and doing a pull request with the fix (I fixed it on my machine already) but there are pull requests from 2013 that haven't been reviewed or merged yet so IDK if it will do any good.

ciozi137, 2021-02-24:
@Monkeymerlot  I had to do a quick security update related to instrument/postgreSQL. I can look at the config stuff now.

Monkeymerlot, 2021-02-24:
So testing for an empty variant will not work.
Without my fix for the variant configuration palette:
[image]
With Fix:
[image]
and test VI:
[image]

ciozi137, 2021-02-24:
@Monkeymerlot thanks for checking. I also just checked and the result was that my config file was erased completely. I am going to take your zip file above and merge

ciozi137, 2021-02-24:
@Monkeymerlot looking good. Update will be included in next build (1.8.3 I think)

Here is my test:
[image]

Monkeymerlot, 2021-02-24:
The test looks good but I think you might have to delete the temp file once youre done with it.

ciozi137, 2021-02-24:
@Monkeymerlot I delete it at the beginning of the test

Monkeymerlot, 2021-02-24:
Oops missed that

Monkeymerlot, 2021-02-24:
I was having some issues getting everything to work because it was looking for the Configuration Prototype file, but it was also not part of the class. So I put it in the class then deleted it from the project and pushed the changes. Hopefully that will fix it.

ciozi137, 2021-02-24:
@Monkeymerlot no I took your code and added it to my VIs. So just update the instrument framework on vipm

Monkeymerlot, 2021-02-24:
@ciozi137 I understand. I'm saying the project was looking for it probably when you had it open to copy the code. Like a ghost dependency. And it was making the class have a broken run arrow. It should be good now though.

Monkeymerlot, 2021-02-24:
@ciozi137 I am not seeing a new package (current package is 1.7.0.95). Were you saying that you wanted me to build the package or did you not get to it yet or is there a problem on my end?

ciozi137, 2021-02-24:
@Monkeymerlot refresh the repo. Latest version is 1.8.3

Monkeymerlot, 2021-02-24:
@ciozi137 That was the issue \*slaps forehead\*. Thanks

ciozi137, 2021-02-25:
@Monkeymerlot I just reverted your commit. Instrument Framework is programmed in LV2016 and you should be installing via VIPM. Can you elaborate what issue you were having with dependencies? I have no issue on my system

ciozi137, 2021-02-25:
@Monkeymerlot sorry I see the problem now. I will build a new version (will be v1.8.3.100)

Monkeymerlot, 2021-02-25:
No worries. LabVIEW just loves bringing us back to dependency hell...

closed 2023-09-29 by ciozi137