levylabpitt/Instrument-Framework #109: Feature Request: Implement IDN Method for Instrument Identification

pgwijesinghe, 2024-06-04:
I propose adding a feature where all instruments output an identification (IDN) dictionary upon request. This feature would be similar to outputting the set of available commands upon a 'HELP' request.

e.g.
```
[In] {"jsonrpc": "2.0", 
            "method": "IDN",
            "id": "id"} 
[Out] {"jsonrpc": "2.0", 
            "result": {'vendor': 'Levylab',
                          'model': 'MC Lockin',
                          'serial': None,
                          'firmware': 'v2.15.4.4'},
            "id": "id"}
```

ciozi137, 2024-06-04:
Good idea. Any ideas how to generate a serial number for a soft instrument? Things like the cryostats are more or less straightforward because they could be derived from the hardware. But we have X lockins that are all identical... maybe I can get a number from the pxi chassis/4461 cards

ciozi137, 2024-06-07:
[image]

`Write IDN` is a public method that the child class will call (where? `onProcessStarted`?) to set the serial number and firmware number

ciozi137, 2024-06-07:
[image]

ciozi137, 2024-06-11:
Protected method to set IDN data:
[image]

ciozi137, 2024-06-12:
[image]

IDN Data Flow:
1. `Child Instrument:process.vi` calls `Open Hardware.vi`
2. `Open Hardware.vi` opens connection to the hardware, and then calls `Write IDN.vi`
3. `Write IDN.vi` queries the S/N (or reads a constant, it's up to the developer) and then calls `Instrument.sendMessageToProcess.vi`, the message is then handled by `Handle Command.vi` in the `Instrument:process.vi`.