levylabpitt/Instrument-Framework #164: Caraya Dependency

ciozi137, 2025-05-03:
Classes should not depend directly on Caraya.

Question: How to use Caraya to test private methods?

ciozi137, 2025-05-03:
If have parent.lvclass:protected.vi and I can make an override parent.test.lvclass:protected.vi to test. But I'm not sure about the case with parent.test.lvclass:private.vi

ciozi137, 2025-05-03:
The answer might be to not do that? Just test the interface methods (public/protected/community)

ciozi137, 2025-05-03:
I keep coming back to "don't do that"

- https://forums.ni.com/t5/Unit-Testing-Group/Unit-test-on-private-class-method/m-p/3382725/highlight/true#M6

Monkeymerlot, 2025-05-03:
@ciozi137 so I generally agree with the sentiment of not testing private methods. I can elaborate on it more, but the idea is that you care about _what_ your code is doing, not _how_ it is doing it. So private methods are generally involved in _how_ your code actually does something, while your public methods are a promise to do something, fulfilling the _what_ part of the above statement. 

The main benefit of not testing your private methods/implementation I think can be explained with the ZMQ stuff you did, where you also have an implementation in STM. What you _really_ care about is that you are able to communicate between two different programs, but whether or not you use ZMQ or STM to do it doesn't really matter. If you only test your public methods, and you started out with STM and later switched it over to ZMQ, your tests will still be valid, and you will be able to detect if your change over to ZMQ breaks your messaging API. 

To get rid of Caraya as a dependency, I will not put tests in the same project as I have the working code in. Instead, I have another project that I call "Test ___________". If I remember correctly, the SMO project does this really well. I'll check on how I do it exactly and put it in detail in another comment.

ciozi137, 2025-05-03:
Indeed, I was looking at the SMO project and see that it is also not testing private methods.

Monkeymerlot, 2025-05-03:
OK so I just use the classes as a dependency in the project, and only access their public API. A good example of how I do this is the ParseTesting.lvproj in the lithography program.

Just another comment now that I took a look at that thread you posted here that I don't think was mentioned there:

If you have a private method that your class relies on in multiple places, that possible changes to it will have down stream effects on multiple public facing methods in the class you have, I am of the opinion that you can put it into a "helper" class or just as on its own, and implement methods to test that helper class separately, because that will then be the "public API" of that new helper class.

ciozi137, 2025-05-04:
### Fixes:
- LV-Data, LV-Data.PGSQL v3.10.0
- Logger.PGSQL v1.2.1
- Remote Control.ZQM v1.2.1
- Instrument Framework (implicit dependency via RemoteControl.ZMQ, etc) v1.15.1

32a4f98...763520b

ciozi137, 2025-05-04:
build time cut from 11 min to 7 min (Instrument Framework)

closed 2025-05-04 by ciozi137